casept/linux-samsung-smartwatch #9: Shutdown doesn't work

casept, 2024-03-16:
The display at least still seems to be active after a shutdown, so we're possibly missing something required to finalize shutdown.

Reboot works fine.

casept, 2024-03-16:
Samsung wiggles a magic PMU register in the downstream kernel at https://github.com/casept/samsung-rinato-sources/blob/master/kernel/arch/arm/mach-exynos/sec_reboot.c#L43 .

The mainline kernel defines this register in `arch/arm/mach-s5pv210/regs-clock.h` as `S5P_INFORM2`, but doesn't use it.

casept, 2024-03-17:
Seems like there's no power-off driver for any Exynos platform in mainline, the PMU driver only implements suspend. Will try writing one.

PabloPL, 2024-03-17:
On Galaxy S1 (s5pv210) which also requires writing specific value to specific register, we did used **syscon-poweroff** driver. Maybe this could be used also in this case. Here is link to dts node https://github.com/torvalds/linux/blob/master/arch/arm/boot/dts/samsung/s5pv210-aries.dtsi#L541

PabloPL, 2024-03-17:
Also that code on https://github.com/casept/samsung-rinato-sources/blob/master/kernel/arch/arm/mach-exynos/sec_reboot.c#L43 - isn't it doing LP charging mode instead of power off which is at https://github.com/casept/samsung-rinato-sources/blob/master/kernel/arch/arm/mach-exynos/sec_reboot.c#L62 ?

PabloPL, 2024-03-17:
Or maybe just include exynos-syscon-restart.dtsi like all other exynos based devices in mainline.

casept, 2024-03-17:
I've implemented a driver now, and power off seems to work. Might be possible to solve this more elegantly by using the syscon-poweroff DT node instead.

closed 2024-03-17 by casept